jp112sdl/AskSinAnalyzer #3: CR: show every single flag in 'Flags' as colored rounded rectangle

jp112sdl, 2019-06-05:
each flag has its own color.
i.e.:
- BCAST: LIGHTGREY
- INFO: BLUE
- ...

psi-4ward, 2019-06-05:
Do we have a list of all possible flags?
Do you prefer specific colors like blue for info?

jp112sdl, 2019-06-05:
_Sorry - INFO was wrong!_

The flags can be
- "WKUP "
- "WKMEUP "
- "BCAST "
- "BURST "
- "BIDI "
- "RPTED "
- "RPTEN "

The flags `RPTED` and `RPTEN` are imho unimportant.
They can be shown in a lightweight color

`BURST` can be red - causes high DC.
`BIDI` maybe orange

`BCAST` is a non-acknowledged message so it can be in a light grey

`WKUP` and `WKMEUP` have informational character. The should not be an eye catcher...

jp112sdl, 2019-06-06:
Only show a small colored rounded rectangle without text, but always on its fixed position.
Show a legend on the top of the page

psi-4ward, 2019-06-06:
Tried it, looks not very good and in my op bad user experience. I'll dense the flags a bit more.

commit 2e747194 (2019-06-05): WebUI: Add flag styling (closes #3)
closed 2019-06-06 by psi-4ward