jp112sdl/AskSinAnalyzer #12: OTA from WebUI

gulalo, 2019-07-01:
Hi Jérôme,
vielen Dank für das super Tool!
Habe OTA versucht über WebUI,scheint aber nicht zu laufen.
von Version 1.3-->1.4
es erscheint: https://raw.githubusercontent.com/jp112sdl/AskSinAnalyzer/master/ota/AskSinAnalyzerESP32.bin
weiter passiert aber nichts, bzw. wie lange kanns dauern?
Habe ich etwas übersehen??

jp112sdl, 2019-07-01:
Hi!
Ich sage mal so ca. 30 Sekunden.

Im seriellen Monitor müsste was mit "Check for Update..." kommen - dann eine ganze Weile nix und dann der Reboot

chkbln, 2019-07-01:
Hi,
bei mir funktioniert das Update auch nicht.
Nach dem Klick auf den den Update Button kommt
https://raw.githubusercontent.com/jp112sdl/AskSinAnalyzer/master/ota/AskSinAnalyzerESP32-ND.bin

[image]

Der ESP ist dann auch nicht mehr über die IP erreichbar.
Nach dem Neustart funktioniert er wieder, aber mit der alter Version 1.3.
Im seriellen Monitor wird "Check for Update..." nicht angezeigt
Gruß

jp112sdl, 2019-07-01:
Es gibt nun noch zahlreiche Möglichkeiten, weshalb es nicht klappt..., zB. wenn du die 1.3 mit der neuesten Version von AsyncTCP und ESP32 Board 1.0.2 kompiliert hattest, was nun zu einem WDT Reset führt (ist mit 1.4 behoben).

Am besten du kompilierst die 1.4 noch mal selbst und lädst sie auf den ESP32

psi-4ward, 2019-07-02:
Was genau heißt: 
> Nach dem Klick auf den den Update Button kommt
https://raw.githubusercontent.com/jp112sdl/AskSinAnalyzer/master/ota/AskSinAnalyzerESP32-ND.bin

Im Browser sollte dann etwas stehen wie 

`http://<espip>/httpupdate?url=https://raw.githubusercontent.com/jp112sdl/AskSinAnalyzer/master/ota/AskSinAnalyzerESP32-ND.bin`

chkbln, 2019-07-02:
@psi-4ward 
ich habe jetzt, wie Jérôme empfohlen hat, die Version 1.4 neu kompiliert und auf den ESP32 geladen.
Kann das Fehlerbild also nicht mehr reproduzieren.

psi-4ward, 2019-07-02:
Na dann machen wir issue mal zu. Sofern wieder Probleme auftauchen gehen wir es noch einmal an. Ich hab heute auch noch mal von 1.3 auf 1.4 das Update gemacht

closed 2019-07-02 by psi-4ward