jp112sdl/AskSinAnalyzer #19: Gerätenamen werden nicht aufgelöst

Xento, 2019-08-08:
Ich habe alles zusammen gebaut und es werden auch Pakete empfangen.
Allerdings steht oben rechts immer 
API Error getLogByLogNumber: Not resolving 634163 due to wrong length

Alle Seriennummern sind bei mir nur 6 Zeichen lang. Im Code wird geprüft ob sie 10 Zeichen lang ist, daher kommt der Fehler.

jp112sdl, 2019-08-08:
Moin!
Kannst du bitte mal im seriellen Monitor schauen (57600 Baud), was beim Empfang eines Telegramms ausgegeben wird?

Mir fallen auf Anhieb 3 mögliche Ursachen ein:
1.) Bei `634163` handelt es sich um ein HmIP Gerät. Derzeit werden im master-Branch nur HM RF (also klassische BidCos Geräte) unterstützt
2.) Das [zusätzliche Programm](https://github.com/jp112sdl/AskSinAnalyzer/wiki/CCU_Unterst%C3%BCtzung) auf der CCU ist falsch oder fehlerhaft.
3.) Firewalleinstellungen auf der CCU verhindern Anfragen an Port 8181

Du kannst auch gern mal die neue Beta-Version testen:
https://homematic-forum.de/forum/viewtopic.php?f=76&t=52117
Bei der werden (so zumindest unsere Absicht ^^) auch HmIP Geräte aufgelöst und auch das Programm ist nicht notwendig.

Xento, 2019-08-08:
Moin, also das 634163 ist ein HM-TC-IT-WM-W-EU:
```
 cd /etc/config/rfd;grep -i "\" address=\"0x634163\"" *.dev|awk -F"." '{print $1}'
OEQ1675602
```
Das Program ist angelegt und der Port 8181 ist auch frei.
Werde mir mal die aktuelle Version hier aus dem dev bauen und drauf spielen.

Xento, 2019-08-08:
Tatsache, hab nun die DEV Version drauf, dort tut nun alles.
Vielleicht sollte man beim Wiki nen Vermerk machen, dass es nicht mehr UpToDate ist.

closed 2019-08-08 by Xento

jp112sdl, 2019-08-08:
>  Vielleicht sollte man beim Wiki nen Vermerk machen, dass es nicht mehr UpToDate ist.

Die neue Version ist erst seit gestern online und noch im BETA (dev) Status. ;)

Xento, 2019-08-08:
Oh dann hab ich ja genau den richtigen Zeitpunkt erwischt xD