jp112sdl/AskSinAnalyzer #27: Baudrate arduino abweichend

mikewolf, 2019-08-12:
Es gibt beim nano 2 bootloade
Welche dann unter 57 kb oder 115k
Auf der seriellen senden eventuell anpassbar bei ota ?

stan23, 2019-08-13:
Beim Analyzer ist ein Arduino Pro Mini im Einsatz, mit 57600 Baud für die serielle Schnittstelle.
OTA ist das eigentlich nicht vorgesehen, die CCU kennt das Gerät ja gar nicht.

jp112sdl, 2019-08-13:
> nano

Bedenke auch, dass der Nano mit 5V Pegeln arbeitet. Damit wird das CC1101 nicht lange überleben.

mikewolf, 2019-08-19:
hab nen nano und der läuft mit57k baudrate, ist 3V daher habe ich den ESP32 angepasst aber nach update ist diese anpassung fort ,,,

jp112sdl, 2019-08-20:
Dann sollte ja alles passen. Hier in diesem Projekt arbeitet alles mit 57600 Baud.

> daher habe ich den ESP32 angepasst aber nach update ist diese anpassung fort

Oder du kaufst dir für 2 EUR einen Pro Mini, um bei den Projektspezifikationen zu bleiben.
Ansonsten darfst du das Update über die Weboberfläche nicht durchführen, sondern musst bei einer neuen Version den master runterladen, deine Anpassung durchführen und dann selbst kompilieren.

closed 2019-09-23 by mikewolf