jp112sdl/AskSinAnalyzer #32: Anzeige des Namens von Serial "HmIP-RF" leider immer noch schwarz

venice1200, 2019-08-16:
Hallo,
habe gerade (2019-08-16/9:00) die aktuellste Version gezogen und installiert.
Leider wird im Web der Name von "HmIP-RF" noch immer schwarz dagestellt.

Auszug aus der Device List:
{"address":12470xxx,"serial":"HmIP-RF","name":"HmIP-CCU"}

[image]

Reference: #28 

Danke :-)

venice1200, 2019-08-16:
Sowohl in der "Von" als auch im "An" Spalte.

[image]

venice1200, 2019-08-16:
Hab mir jetzt erst einmal so geholfen indem ich im HM Programm 7 Freizeichen an den String "HmIP-RF" angehängt habe damit der String (die Serial) 14 Zeichen lang ist Somit ist "isIP" erfüllt.

if ((HmIPCentral > 0) && (data == HmIPCentral)) { ser = "HmIP-RF       "; }

VG
Lars

psi-4ward, 2019-08-16:
Lass mich wissen wenns noch Probleme gibt ;)

commit c1b9d7c1 (2019-08-16): Fix: HmIP-RF not blue (closes #32)
closed 2019-08-16 by psi-4ward