jp112sdl/AskSinAnalyzer #37: AskSinAnalyzerESP32.ino

audi2010, 2019-08-21:
Hallo Jerome,
wo kommt den die Datei Zeile 15 "<Preferences.h>" her.
Die fehlt mir.

jp112sdl, 2019-08-21:
Hi!

Was meinst du mit "wo kommt die her?"

Sie ist Teil des ESP32-Boards und wird automatisch mit ausgeliefert.
https://github.com/espressif/arduino-esp32/tree/master/libraries/Preferences/src

psi-4ward, 2019-08-21:
Du hast wahrscheinlich einfach die falsche Lib erwischt.
Ist mir auch passiert. Kannst du nicht einfach das Hex flashen oder hast du Anpassungen vorgenommen?

Alternativ zur Arduino IDE könntest du PlatformIO verwenden, dem Ordner liegt eine `platformio.ini` bei welche alle nötigen Libs referenziert. Es reicht in dem Fall ein `platformio run -t upload` und es wird alle nötige heruntergeladen, compiliert und geflasht.

jp112sdl, 2019-08-21:
> Du hast wahrscheinlich einfach die falsche Lib erwischt.

Wie schafft man das?

audi2010, 2019-08-21:
Ich schaff alles :)
Auf einem anderen Rechner läufts problemlos.
Danke für die Tips

jp112sdl, 2019-09-01:
Hier kann wohl zu

closed 2019-09-01 by jp112sdl