jp112sdl/AskSinAnalyzer #39: WebUI: Ändern von Konfigurationsdaten

jp112sdl, 2019-08-27:
Bitte eine Konfigurationsseite bereitstellen, um direkt aus der WebUI Parameter ändern zu können.
Der Request lautet `/setConfig` und nimmt POST entgegen.

Folgende Parameter können gesetzt werden:
- Statische IP-Konfiguration (`ip`, `netmask`, `gw`)
- CCU IP (`ccuip`)
- NTP Server (`ntp`)
- Hostname (`hostname`)

Beispiel:
 `curl -d ccuip=192.168.1.252 http://192.168.1.199/setConfig`

psi-4ward, 2020-02-13:
Hab mal was in Dev-Branch gepusht. Hab leider meinen Analyzer nicht da zum direkt testen :\

jp112sdl, 2020-02-13:
Sieht schon gut aus!

Bitte noch ergänzen:

- Backend (`backend`) = Auswahl "CCU" oder "Andere"
  - wenn "CCU", dann Feld "CCU IP" sichtbar
  - wenn "Andere", dann Feld "Backend URL" sichtbar
- Backend-URL (`backendurl`) = URL, wenn Geräteliste aus anderer Quelle geladen werden soll
- RSSI Histogram Bar Width (`rssi_hbw`) = Breite der RSSI Histogramm-Balken auf dem TFT (zulässig > 0 bis < 64)
- RSSI Alarm Threshold (`rssi_alarmthreshold`) = Schwellwert für RSSI Alarmmeldung in der CCU
- RSSI Alarm Count (`rssi_alarmcount`) = Anzahl an stetigen Schwellwertüberschreitungen, bevor in der CCU der Alarm ausgelöst wird

jp112sdl, 2020-02-14:
Perfekt, danke!

closed 2020-02-14 by jp112sdl