jp112sdl/AskSinAnalyzer #45: WPS-Kopplung implementieren

t0mx, 2019-10-24:
Eingabe des WLAN-Passworts im AP-Modus des ESP32 ist unsicher. Auch wenn die Wahrscheinlichkeit, dass es im richtigen Moment aufgeschnappt wird, recht gering ist, bleibt es eine Sicherheitslücke, die durch den Einsatz von WPS geschlossen werden könnte. Zumindest für diejenigen, die sich des Risikos bewusst sind.

jp112sdl, 2019-10-25:
Zum Thema WPS:
https://www.tecchannel.de/a/netzwerk-in-gefahr-wie-angreifer-wlans-hacken,2039709,4
https://blog.avira.com/de/wi-fi-protected-setup-wps-sicherheitsrisiko/

"Auch wenn die Wahrscheinlichkeit, ..., recht gering ist..." Willst du trotzdem noch WPS nutzen? ^^

Wie im Forum bereits geschrieben, werde ich da nichts machen.

Aber ich lass das mal offen, vielleicht hat ein anderer mal Lust und Laune, das zu implementieren

stan23, 2019-10-25:
Kann man das Problem vielleicht ganz anders lösen?
In dem man die Werte direkt in den Speicher schreibt? Bietet eines der ESP-Flasher-Tools einen Editor für den Speicher?
Oder jemand implementiert die Touch-Unterstützung und einen Bildschirmtastatur.

jp112sdl, 2019-10-25:
> In dem man die Werte direkt in den Speicher schreibt? Bietet eines der ESP-Flasher-Tools einen Editor für den Speicher?

Da kenne ich mich nicht aus. Hab mich mit den Internas des NVS nie befasst.
Ansatz wäre wohl die [Preferences](https://github.com/espressif/arduino-esp32/tree/master/libraries/Preferences/src)-Implementierung im Detail zu betrachten, bzw. der dabei verwendeten [nvs.h](https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/nvs_flash/nvs.h).

> Oder jemand implementiert die Touch-Unterstützung und einen Bildschirmtastatur.

Vergebene Müh'... Dann kommt einer mit "Ich nutze den Analyser ohne Display." 🤓

t0mx, 2019-10-25:
Die in den Links beschriebene Sicherheitslücke bezieht sich, soweit ich weiß auf die WPS-Methode mit PIN. Über die andere Methode konnte ich nichts dergleichen finden, sollte sie sich als genau so unsicher erweisen, wäre das natürlich auch vergebene Mühe.
Der Aufwand scheint überschaubar zu sein: [https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WPS/WPS.ino](url), aber ich wüsste nicht, wo ich das Ganze einfügen kann, insbesondere müsste man ja auch eine Möglichkeit zum Anstoßen in der WebUI vorsehen.

jp112sdl, 2019-10-25:
> Der Aufwand scheint überschaubar zu sein 

Na dann... Feuer frei!

> aber ich wüsste nicht, wo ich das Ganze einfügen kann, insbesondere müsste man ja auch eine Möglichkeit zum Anstoßen in der WebUI vorsehen.

Das (beides) muss in den WiFiManager rein, der ist in der [WM.h](https://github.com/jp112sdl/AskSinAnalyzer/blob/master/AskSinAnalyzerESP32/WM.h) sowie [WM.cpp](https://github.com/jp112sdl/AskSinAnalyzer/blob/master/AskSinAnalyzerESP32/WM.cpp)
 zu finden.
Dort stehen für den ESP32 WPS Kram auch schon `TODO` marker.

Der WiFiManager kommt ursprünglich von hier:
https://github.com/tzapu/WiFiManager

Evtl. ist da auch WPS jetzt schon drin und du kannst es dir von dort rüberholen.

closed 2020-02-01 by jp112sdl