jp112sdl/AskSinAnalyzer #47: ESP32 Belegung

tis-cs, 2019-12-21:
Moin,
ich habe hier leider ein ESP32 Board mit jeweils nur 15 Pinnen pro Seite. Wäre es möglich, dass die Anleitung der Verdrahtung diesbezüglich nicht nur ein Bild zeigt sondern auch die Pinbeschreibung enthält?

Danke im Voraus!

stan23, 2019-12-22:
Servus,
du könntest dich an den Pinnamen hier im PDF orientieren:
https://github.com/stan23/myPCBs/blob/master/AskSinAnalyzer/AskSinAnalyzer_V2.0_Schematic.pdf

Die Pins zum Touch-Controller (T_*) sind optional und nicht von der SW unterstützt.

tis-cs, 2019-12-23:
@stan23 Vielen Dank für den Hinweis. Allerdings bekomme ich aus dem Dokument keinen Zusammenhang bzw. eine Zuordnung zu meinem ESP32 hin. Verstehe ich etwas grundlegendes nicht oder was mache ich falsch? 

Könntest du mir ein zwei Beispiele anhand meines Bildes machen (z.B. Display zum ESP)

Vielen lieben Dank!

jp112sdl, 2019-12-23:
Im Sketch findest du die Pins, wie sie auch auf deinem Board stehen:
```
#define TFT_LED                 33
#define TFT_CS                   5
#define TFT_RST                 26
#define TFT_DC                  25
```
MISO ist Pin19 
MOSI ist Pin23

stan23, 2019-12-23:
Wenn du die wiring.fzz mit Fritzing öffnest, siehst du die Pinnamen.

Ich hoffe ich habe alle richtig übertragen:
[image]

tis-cs, 2019-12-24:
Super, danke euch beiden für die Hilfe.

closed 2019-12-24 by tis-cs