jp112sdl/AskSinAnalyzer #48: Importieren von LOG in die WEB nicht möglich...

ThomasLoeffel, 2019-12-27:
Hallo Zusammen,

danke für Eure "SUPER" Arbeit! Folgendes ist mir aufgefallen.
Im WEBUI beim Start bekomme ich bei mir max 1h angezeigt. Somit dachte ich das ich nach einem CSV download (25.12 bis dato) und Import der gleichen Datei zurück mehr angezeigt bekomme. Der Import startet aber augenscheinlich nicht. Nur Bereich Telegramme pro Device wird "weiss" aber sonst passiert nichts. Mache ich was falsch?

Lieben Gruß Thomas

[image]

[image]

jp112sdl, 2020-02-14:
Hallo Thomas, tritt das Problem immer noch auf?

ThomasLoeffel, 2020-02-14:
Hallo Jerome,

ich verwende die Version 3.2 (gerade heraus gekommen). Ja der Fehler ist immer noch so. Ich habe es selbst mit kleinsten Dateien versucht. 1 min Asksin laufen lassen. CSV Download "Name: log.csv". Dann Asksin neu starten und Import der Datei"log.csv". der blaue Button blinkt einmal, das Kreisdiagramm ist leer. Aber das Importfenster verschwindet nicht. Sonst keine Reaktion bis ich das Browserfenster aktualisiere. Dann alles wie immer, aber ohne log.csv Daten. Ich benutze Firefox.

Danke für Deine super Arbeit und hoffe auf eine Aufklärung :-).
Gruß Thomas

jp112sdl, 2020-02-14:
Kannst du mal eine CSV hier anhängen, um das Problem evtl. reproduzieren zu können?

ThomasLoeffel, 2020-02-14:
[log.txt](https://github.com/jp112sdl/AskSinAnalyzer/files/4205948/log.txt)

psi-4ward, 2020-02-14:
Cool. Ich schau mir das am Sonntag an. Welchen Browser auf welchem OS hast du?

ThomasLoeffel, 2020-02-14:
System: Win10 pro, Browser: Firefox 73.0 (64-Bit), komischerweise lässt das WebUi mit MS Edge nicht anzeigen.

ThomasLoeffel, 2020-02-14:
Nur am Rande erwähnt. Mit der Version 3.2 funktioniert ein Reboot bei mit nicht mehr zuverlässig. Das bedeutet das in 6 von 10 Fällen der Asksin nicht mehr hochfährt (kleine grüne Punkt wird nicht voll). Dann hilft nur Strom aus/an.

ThomasLoeffel, 2020-02-14:
Zusatz: Kann aber auch an dem Problem mit dem Import liegen.... Habe eben ohne Import zu nutzen den Reboot probiert und da ist es "0" mal passiert.

closed 2020-02-14 by psi-4ward

psi-4ward, 2020-02-14:
Sooo teste mal bitte ;)
CSV + (dummer blöder) Edge sollte nun gehen

ThomasLoeffel, 2020-02-14:
Mal eine dumme abschließende Frage:
Da Du keine neue Version vergeben hast, muss ich dann selbst compilieren und mit Arduino IDE drauf spielen? Oder gibt es einen einfacheren Weg?

Gruß und Danke !!
Thomas

psi-4ward, 2020-02-14:
Nein, die Version bezieht sich auf die Software auf dem ESP. Die WebUI wird aus der _Cloud_ (von Github) gezogen und sollte demnach aktuell sein. Ggf. musst du den Browser-Cache löschen.

ThomasLoeffel, 2020-02-14:
Hat super geklappt :-). Geht alles wie von Dir erwartet. 
Idee: Kann man ggf die ESP-Version und die UI-Version auf dem Display (auf dem ESP) auf der Seite mit den Netzwerkdaten hinterlegen? 

Lieben Gruß und vielen Dank Thomas !!!

commit 81db7905 (2020-02-14): Fix CSV Import (closes #48)