jp112sdl/AskSinAnalyzer #50: Doku-Erweiterung: Was geht, was geht nicht?

mikes222, 2020-01-04:
Bevor ich anfange, das nachzubasteln würde ich gerne wissen, ob das Gerät auch mit dem IP Protokoll umgehen kann oder nur mit dem "alten" Protokoll der HM-Geräte. 
Weiters: Was ist, wenn die Pakete verschlüsselt sind? Was sieht man dann? 
Vielen Dank für Eure Aufklärung und eventuell Anpassung des Wikis.

jp112sdl, 2020-01-04:
Von der HmIP Kommunikation siehst du lediglich Sender und Empfänger, RSSI (am Analyzer gemessen) und Telegrammlänge (Bytes).

Telegramm-Counter, -Typ und -Flags werden nicht angezeigt.

closed 2020-01-04 by mikes222

jp112sdl, 2020-01-04:
P.S.: Beim klassischen HM gibts keine Verschlüsselung der Payload. 
Wie das bei HmIP ist, weiß ich jedoch nicht