jp112sdl/AskSinAnalyzer #57: Datenspeicherung

howy075, 2020-02-17:
Wäre es möglich, dass der Analyzer täglich eine neue .csv erstellt statt alles in eine Datei zu schreiben?

jp112sdl, 2020-02-18:
An sich ne gute Idee. 
Jedoch würde man dann beim Klick auf "CSV Download" nur die Tages-CSV bekommen...

Aber ich glaub, das ist vertretbar. 
Wenn man mehr braucht, muss man entweder die SD Karte entnehmen bzw ich bau noch eine Download-URL mit ein, der man das Tagesdatum übergibt

jp112sdl, 2020-02-18:
Bitte testen. Änderung ist im [dev](https://github.com/jp112sdl/AskSinAnalyzer/tree/dev)-Branch.

"CSV Download" im Web lädt die CSV des aktuellen Tages herunter.
Möchte man einen anderen Tag, so kann man folgende URL aufrufen:
`http://asksinanalyzer.local/download?filename=/yyyymmdd.csv`
- yyyy = Jahr
- mm = Monat
- dd = Tag

psi-4ward, 2020-02-18:
An dieser Stelle wäre noch ein api endpoint sinnvoll welcher eine Liste an Dateien zurück gibt (json array).

Aber ist Tag nicht etwas fein aufgelöst?

jp112sdl, 2020-02-18:
> An dieser Stelle wäre noch ein api endpoint sinnvoll welcher eine Liste an Dateien zurück gibt (json array).

An sich ne schicke Idee, dann könnte man die vorhanden Files als Dropdown auflisten und das gewünschte runterladen.
Ich wollt da jetzt aber nicht noch viel Energie reinstecken, vor allem was die WebUI betrifft.

> Aber ist Tag nicht etwas fein aufgelöst?

Kommt sich drauf an, wie viel anfällt.
Bei mir ca. 15k Zeilen pro Tag. Täglich ist schon schick. Monatlich wäre zu lange und wöchentlich ist ein blöder Rhythmus. :)

jp112sdl, 2020-02-18:
Hab das Directory Listing mal fix mit eingebaut.
Hängt im `/getConfig` mit an, kann auch manuell mit `/listSD` abgefragt werden.

[image]

Was damit nun passiert, kannst du (@psi-4ward) dir überlegen 🤓

howy075, 2020-02-18:
Gefällt mir sehr gut so!
Vielen Dank!

howy075, 2020-02-18:
Jetzt könnte man evtl darüber nachdenken, ob die webui automatisch die Daten des aktuellen Tages ladet.
Aber das ist wirklich ein Wunsch der gerne unerfüllt bleibt! sind ja eh nur 2-3 klicks.
Grüße
Howy

jp112sdl, 2020-02-19:
Das wird zu viel... aber wenn man den ganzen Tag haben will, kann man ja auch die CSV uploaden

closed 2020-02-19 by jp112sdl