jp112sdl/AskSinAnalyzer #59: RSSI Alarm Threshold

friedpa, 2020-03-25:
Wenn man den RSSI Alarm Threshold aktiviert und den ASA mit einer (RasberryMatic) RM verbunden hat, kommt nach einem Reset vom ASA immer einmal ein Alarm an der RM an, egal welchen Pegel man eingestellt hat und auch der RSSI Alarm Count hat keinen Einfluss darauf. Ein durchforsten aller Telegramme in der Liste zeigt auch keinen Wert der über dem Threshold liegt an.

jp112sdl, 2020-03-25:
Hmm, kann ich leider nicht reproduzieren.

Kannst du mal schauen, was der ESP32 an seriellen Ausgaben beim Booten schreibt?

friedpa, 2020-03-26:
Guten Morgen, bin leider kein Profi so wie Ihr, was genau soll ich machen?

jp112sdl, 2020-03-26:
Den seriellen Monitor in der Arduino IDE öffnen, 57600 Baud einstellen, dann Reset-Taster am ESP32 drücken

friedpa, 2020-03-26:
Muss ich sonst noch was umstellen? Wenn ich den ASA an den PC über das USB Kabel anschliesse sehe ich den ASA nicht.

friedpa, 2020-03-26:
Screenshots:

[image]
[image]
[image]

jp112sdl, 2020-03-26:
> Wenn ich den ASA an den PC über das USB Kabel anschliesse sehe ich den ASA nicht.

Wie hast du ursprünglich mal den Sketch geflasht?

jp112sdl, 2020-03-26:
Der Alarmthreshold mit -120 ist verdammt gering... Mach mal bitte -100 und Alarmcount auf 60

friedpa, 2020-03-26:
Ich habe das Gerät fertig bekommen :o)
Probiere -100 und 60 melde mich

friedpa, 2020-03-26:
Nunn mit -100dB und 60

[image]

friedpa, 2020-03-26:
Mit -100dB und 60 das selbe Problem

jp112sdl, 2020-03-26:
Ich seh da in dem Screenshot vorhin gar keine rote RSSI-Linie... ich vermute mal, die ganze Problematik kommt, weil auf dem AskSinSniffer328P noch die alte Firmware läuft.

friedpa, 2020-03-26:
Ok, das heißt ich muss diese FW updaten .... muss also irgendwie nun den ASA an den PC bekommen .... danke für Deinen Support

stan23, 2020-03-26:
Du brauchst einen FTDI-Adapter...

friedpa, 2020-03-26:
OK, danke habe ich... gibt es irgendwo eine Anleitung? Sorry, wenn ich Fragen stelle die für Euch eh klar sind ....

jp112sdl, 2020-03-26:
Oder dort nachfragen, wo du den AskSinAnalyzer "bezogen" hast... 
Gerät noch mal hinschicken, updaten lassen

friedpa, 2020-03-26:
Gents, noch eine Frage: ist das Ding überhaupt zum Upgraden?

[image]

stan23, 2020-03-26:
Ja ist es, das Platinendesign stammt von mir, Idee und Schaltplan von Jérôme.
Der FTDI muss am J21 um den ATmega zu aktualisieren.

friedpa, 2020-03-26:
Dann habe ich das Ding eh von Dir :o) Du hast damals mehrere Platinen in China bestellt und da durfte ich mitbestellen :o) D.h. ich löte um J21 eine Stiftreihe an und mit einem FTDI Adapter dann das Upgrade.... Gibt es dazu vielleicht eine Anleitung :o)

psi-4ward, 2020-03-26:
Nichts einlöten, steck den FTDI einfach so rein, das sollte ausreichen.
Anleitung gemäß https://github.com/jp112sdl/AskSinAnalyzer/tree/master/AskSinSniffer328P

Siehe auch https://asksinpp.de

friedpa, 2020-03-26:
Super! Danke! Erst jetzt sehe ich, dass da ja stan23 auf der Platine steht, also chatte ich mit den Chefs persönlich :o) Nochmals vielen Dank für das Super Projekt!!!

friedpa, 2020-03-26:
Habe jetzt die FW upgedatet und der Alarm kommt jetzt nicht mehr. Vielen Dank für den Hint und die guidance durch die Problemstellung.

closed 2020-03-26 by friedpa