jp112sdl/AskSinAnalyzer #60: Darstellung der RSSI Linie

friedpa, 2020-04-09:
Ich habe eine Frage zu der Darstellung der roten RSSI Linie. Wenn ich den ASA einschalte, zeichnet er mir die RSSI Linie auf, aber nur solange bis der Zoom von 1m auf 5m umspringt. Dann ist es vorbei mit der roten Linie :o) Das ganze passiert auf einem Firefox Browser (letzte Version) und einem IE.

[image]

Liebe Grüße
Paul

psi-4ward, 2020-04-09:
Die sollte natürlich weiter laufen. Hast du den Chrome probiert? Steht was in der developer console (F12)?

friedpa, 2020-04-09:
Chrome habe ich keinen installiert .... zu resourcenintensiv.
bei der Developer Console musst Du mir helfen. Von welchem Reiter soll ich einen Screenshot machen?
Hier einmal der Inspektor:

[image]

psi-4ward, 2020-04-09:
Einfach mal im Reiter "Konsole" kucken ob was rot wird.

friedpa, 2020-04-09:
Ergebnis:

[image]

psi-4ward, 2020-04-09:
Das sagt mir ja mal gar nichts. Wenn du es auf 1min lässt läuft es aber als weiter?

friedpa, 2020-04-09:
Nope, sobald der 5m Tab oben links auf aktiv geht ist es vorbei mit der roten Linie.

psi-4ward, 2020-04-09:
Ich kann das leider nicht nachstellen. 
Welche Version vom Firefox nutzt du? ESP und AVR Software ist aktuell?

psi-4ward, 2020-04-09:
Wie verhält es sich wenn du nach ca 15 Minuten die WebUI neu aufrufst?

friedpa, 2020-04-10:
Zur Frage 1: Firefox 75.0 (64-Bit)
Frage 2: Firefox stellt die rote Linie wieder dar:
[image]
Der Fehler tritt nur beim erstmaligen "befüllen" des ASA auf

psi-4ward, 2020-04-15:
Das Problem lässt sich nicht nachstellen. Hast du die aktuellen Versionen auf dem avr und esp?

jp112sdl, 2020-04-15:
Habe selben Browser (auf macOS) verwendet.
Meine RSSI Linie bleibt dauerhaft erhalten.

_Habe es jedoch nicht über mehrere Stunden getestet_

friedpa, 2020-04-15:
Ich habe die aktuellen Versionen:

[image]

Wenn sonst niemand das Problem hat wird es wohl an meinem Setup liegen. Wie schon gesgt, die rote Linie wird nicht mehr dargestellt wenn der 5m Button oben links aktiv wird.

jp112sdl, 2020-04-15:
> wenn der 5m Button oben links aktiv wird.

[image]

friedpa, 2020-04-15:
@jp112sdl Warum ist die rote Linie bei Dir nicht über die ganze Breite des Diagrams?

jp112sdl, 2020-04-15:
Das ist by Design.
Der Analyzer hält die letzten 1000 Telegramme vor, sowie die letzten 255 RSSI Messungen.

Da RSSI alle 750ms übertragen wird, hat man beim **Neu-Aufruf** der WebUI nur rund 3 Minuten (255 * 0,75 Sekunden) rückwirkend die RSSI Linie.

Heißt in meinem Bild: Ich habe die WebUI so ca. 11:49:45 Uhr geladen.

friedpa, 2020-04-15:
Darf ich hiezu noch eine Frage stellen: Warum habe ich eine rote Linie von 14:32h bis 15:37h also 65 Minuten und dann stoppt diese?

[image]

jp112sdl, 2020-06-16:
Gibt es derzeit noch Probleme mit der RSSI Linie oder kann das Ticket zu?

friedpa, 2020-06-17:
Hat sich nichts getan, Fehler ist immer noch da:

[image]

Verwende als Browser Firefox

jp112sdl, 2020-06-17:
Komisch, dass das nur bei dir so ist :/