jp112sdl/AskSinAnalyzer #64: Empfänger in Webfrontend / Display stimmt nicht überein

mdk2412, 2020-04-30:
Hallo,

seit kurzem fällt mir auf, dass laut Webfrontend einige meiner Sensoren angeblich an eine Verknüpfung senden (die jedoch nur mehrere Jalousien/Rollläden verknüpft, die Sensoren sind da nicht drin). Im Display steht bei dem gleichen Telegram allerdings richtigerweise als Empfänger "Zentrale". Ist das ein Fehler auf meiner Seite oder im Code?

jp112sdl, 2020-04-30:
Mir fällt da pauschal nix zu ein

mdk2412, 2020-04-30:
Ich hab jetzt mal den Namen der Verknüpfung geändert und die Device List neu geladen. Statt Zentrale kommt dann der geänderte Name der Verknüpfung. Mit anderem Browser/auf anderem Gerät sieht das genauso aus. Wo wird denn aufgelöst, welches Gerät die Zentrale ist? An der Stelle scheint es im Webfrontend schief zu gehen. Auf dem Display ist ja alles korrekt.

psi-4ward, 2020-04-30:
Das passiert in dieser Gegend: https://github.com/jp112sdl/AskSinAnalyzer/blob/master/ui/src/EspService.js#L194

Also versteh ich dich richtig, deine Sensoren schicken laut WebUI Telegramme an die Zentrale aber das ist nicht richtig?

mdk2412, 2020-04-30:
Nein, es ist so: im Webfrontend wird bei jedem Telegramm sowohl bei "AN" als auch "VON" der Name einer Verknüpfung XY angezeigt anstelle des zu erwartenden "ZENTRALE". Im Analyzer-Display wird bei dem identischen Telegramm korrekt "ZENTRALE" angezeigt. Wenn man sich statt der Namen nur die IDs anzeigen lässt, steht da, wo der Name der Verknüpfung angezeigt wird, die richtige ID der CCU. Es wird also der Name der CCU falsch aufgelöst, aber das nur im Webfrontend, nicht im Display.

psi-4ward, 2020-04-30:
Lass da mal bei Gelegenheit noch mal drüber reden

jp112sdl, 2021-01-04:
> Lass da mal bei Gelegenheit noch mal drüber reden

Ich glaub nicht, dass sich diese Gelegenheit noch ergeben wird ;)

closed 2021-01-04 by jp112sdl