jp112sdl/AskSinAnalyzer #66: OTA Update funktioniert nicht

Xento, 2020-05-24:
Hallo,
ich nutze aktuell die Version 3.2 auf meinem ESP32.
Wenn ich versuche auf die Version 3.4 zu aktualisieren, dann erhalte ich folgende Meldung:

`Processing update from https://raw.githubusercontent.com/jp112sdl/AskSinAnalyzer/master/ota/AskSinAnalyzerESP32.bin
Please be patient - ESP32 will reboot automatically`

Danach passiert allerdings nichts mehr un auch wenn ich nach längerer Wartezeit die Startseite aufrufe, ist der ESP zwar online aber immer noch mit der alten Version.

Als Schönheitsfix könnte man auch den Link bei der Warnung, dass der ESP veraltet ist, direkt auf die Info-Seite setzen, da man von dort ja auch das Update startet.

jp112sdl, 2020-05-24:
Ich kenne das Verhalten. 
Es tritt sporadisch und eher selten auf. 
Manchmal schlägt der Download fehl (SSL Handshake) oder es kommt zu einem Timeout.
Blöd ist, dass ich es nicht provozieren kann.

Alternativ bleibt, selbst zu kompilieren und den ESP32 zu flashen.
Oder, falls du einen lokalen Webserver betreibst, diesen als Updateserver zu benutzen.
Du stellst die entsprechende .bin (-ND ist die Version ohne TFT Support) aus https://github.com/jp112sdl/AskSinAnalyzer/tree/master/ota auf deinem Webserver bereit.

Dann startest du das Update mit
`http://asksinanalyzer.local//httpupdate?url=http://deinWebserver/pfad/AskSinAnalyzerESP32.bin`

Xento, 2020-05-24:
Werd ich morgen mal versuchen.
Könnte man nicht etwas mehr Debug Ausgaben einbauen oderso um den Fehler einzugrenzen?

stan23, 2020-05-24:
Die Debug-Ausgaben kommen auf der seriellen Schnittstelle des ESP.
Du kannst dort ja mal einen FTDI-Adapter anstecken und schauen.

jp112sdl, 2020-05-24:
Seriell kommt da eigentlich Debug.

EDIT: Jetzt war Marco schneller 👍

closed 2021-01-04 by jp112sdl