jp112sdl/AskSinAnalyzer #68: Asksin Analyzer ESP32 Software kompilieren

JurgenO, 2020-06-29:
Hallo,
zunächst einmal: vielen Dank für das Bereitstellen des Schaltplans, der Software und der umfangreichen Dokumentation. 

Ich konnte die ESP32 Software nicht direkt kompilieren, sondern die folgende Änderungen waren notwendig. 

1. Füge #include <Arduino.h> an den Anfang von AskSinAnalyzerESP32.cpp
2. In den Libraries Time.h umbenennen in _Time.h
(Verzeichnis: .pio\libdeps\nodemcu-32s\Time_ID44
, siehe https://github.com/esp8266/Arduino/issues/2341)
3. Verwendung spezifischer Library Versionen:

```
lib_deps =
 ESP Async WebServer@1.2.0  # funktioniert !
 ESP32httpUpdate@2.1.145 
 Time@1.6 
 ArduinoJson@6.10.0 
 Adafruit GFX Library
 571 # Adafruit ILI9341
 U8g2_for_Adafruit_GFX
```

Wenn ich <Arduino.h> nicht einfüge führt es zu einem Compilerfehler:
```
Compiling .pio\build\nodemcu-32s\src\AskSinAnalyzerESP32.cpp.o
src\AskSinAnalyzerESP32.cpp:13:7: error: 'String' does not name a type
const String CCU_SV_DEVLIST = "AskSinAnalyzerDevList";  //name of the used system variable on the CCU containing the device list
```

Das Einfügen von Arduino.h führt zu einem Konflikt mit Time.h, deshalb wird es in _Time.h umbenannt.

Ich frage mich warum im Original Source Code Arduino.h nicht inkludiert werden musste.
Gibt es einen eleganteren Weg das Problem zu lösen ?

Gruss vom Niederrhein
Jürgen

jp112sdl, 2020-06-29:
Hi,

leider kenne ich mich mit PlatformIO nicht aus.

Bei Verwendung der Arduino IDE oder Sloeber sind die Anpassungen nicht notwendig.

@psi-4ward kannst du was dazu sagen?
Du hattest damals die `platformio.ini` beigesteuert.

psi-4ward, 2020-06-29:
Hab mir eben mal den _master_ gezogen und mit pio gebaut. Das funktioniert soweit allerdings würde ein flashen nicht funktionieren da `Error: The program size (1398961 bytes) is greater than maximum allowed (1310720 bytes)`

Die aktuellen Versionen scheinen wohl etwas dicker geworden zu sein. Wie groß ist denn die Data-Partition? Diese müsste man wohl noch bei PIO anpassen. 
Ich muss gestehen, so tief bin ich in dem Thema auch nicht drin.

JurgenO, 2020-06-30:
Hallo,

danke für eure schnelle Antwort. 
Ich vermute, dass das Problem an der Entwicklungsumgebung liegt.
Entwickelt Jérôme auf Mac mit der Arduino IDE ? Entwickelt Christoph unter Linux mit PlatformIO ?

Meine Entwicklungsumgebungen sind
(a) Win 10 + Visual Studio Code + PlatformIO
(b) Mac  + Visual Studio Code + PlatformIO

Problem 1: "String" wurde nicht gefunden
---------------------------------------------
Das Problem tritt bei mir in beiden Entwicklungsumgebungen auf. Ich konnte es letztendlich lösen, indem ich  

```
const String CCU_SV_DEVLIST = "AskSinAnalyzerDevList";  //name of the used system variable on the CCU containing the device list
const String CCU_SV_ALARM   = "AskSinAnalyzerAlarm";  //name of the used system variable on the CCU for alarms

```
**nach** den #includes verschoben habe. 
Das inkludieren von Arduino.h ist dann nicht mehr notwendig.

Problem 2: Das Problem mit Time.h
---------------------------------------
Dieses Problem tritt nur in der Windowsumgebung auf. Es kommt zu folgender Fehlermeldung:
```
.pio\libdeps\nodemcu-32s\ESP Async WebServer_ID306\src\WebHandlers.cpp: In member function 'AsyncStaticWebHandler& AsyncStaticWebHandler::setLastModified(tm*)':
.pio\libdeps\nodemcu-32s\ESP Async WebServer_ID306\src\WebHandlers.cpp:67:64: error: 'strftime' was not declared in this scope
   strftime (result,30,"%a, %d %b %Y %H:%M:%S %Z", last_modified);
```

Das Problem ist bereits hier [beschrieben](https://github.com/me-no-dev/ESPAsyncWebServer/issues/96) . Eine Lösung für den AskSin Analyzer ist das Umbenennen der  Include Datei Time.h nach _Time.h in der Time_ID44 Library.

Problem 3: Datei zu groß
---------------------------
Wird bei mir durch die folgende Wahl der Libraries gelöst:

```
lib_deps =
 ESP Async WebServer@1.2.0 
 ESP32httpUpdate@2.1.145 
 Time@1.6 
 ArduinoJson@6.10.0 
 Adafruit GFX Library
 571 # Adafruit ILI9341
 U8g2_for_Adafruit_GFX

```

Habt ihr Interesse diese Änderungen als Branch zu bekommen ? Ich habe noch keine Erfahrung mit GitHub :-)

Danke für eure Unterstützung !

Gruß vom Niederrhein
Jürgen

stan23, 2020-06-30:
> ## Problem 2: Das Problem mit Time.h
>
> Dieses Problem tritt nur in der Windowsumgebung auf. Es kommt zu folgender Fehlermeldung:
>
> ```
> .pio\libdeps\nodemcu-32s\ESP Async [...]

Diesen Fehler kann ich nachstellen.
Meine Entwicklungsumgebung ist  Win 10 + Visual Studio Code 1.46.1 + PlatformIO 4.3.4

Die anderen beiden treten bei mir nicht auf.

```
RAM:   [==        ]  17.8% (used 58440 bytes from 327680 bytes)
Flash: [========= ]  92.3% (used 1209218 bytes from 1310720 bytes)

Dependency Graph
|-- <ESP Async WebServer> 1.2.3 #f13685e
|-- <Time> 1.6
|-- <ArduinoJson> 6.15.2
|-- <Adafruit GFX Library> 1.8.4
|-- <Adafruit ILI9341> 1.5.5
|-- <U8g2_for_Adafruit_GFX> 1.7.0
|-- <ESP32httpUpdate> 2.1.145
|-- <ESPmDNS> 1.0

Espressif 32 1.12.3
```

jp112sdl, 2020-06-30:
Meine Dev-Env sind `macOS + Arduino IDE` bzw. `macOS + Sloeber`.
Da läuft es halt so wie der Code hier im Repo ist.

Macht mir sonst gern nen PullRequest, wenn es Probleme bei PIO gibt, die damit gelöst werden

JurgenO, 2020-07-12:
Hallo,

endlich wieder Zeit für das Projekt :-) . 

Ich kann keinen Branch erzeugen - kann mich bitte jemand als "Collaborator" eintragen ? 

Gruss
Juergen

psi-4ward, 2020-07-12:
Forken und pull request erstellen :-)

JurgenO, 2020-07-12:
Fertig. 
Das ist mein erster Pull Request - ist das so ok ? 

Gruss
Juergen

jp112sdl, 2020-07-12:
Also für mich ergeben sich keine Auswirkungen... habs gemerged

JurgenO, 2020-07-12:
Danke für's mergen.
Es wäre hilfreich das Wiki zu aktualisieren - und zwar hier:
https://github.com/jp112sdl/AskSinAnalyzer/wiki/Arduino_IDE

Kann ich dafür einen Branch erstellen ?

Mein Vorschlag:

Nach (1) und (2):

(3) Für Windows IDE 
- Die Bibliothek Time lokal installieren
- Die Datei "Time.h" umbenennen nach "_Time.h", siehe https://github.com/esp8266/Arduino/issues/2341

Gruss
Juergen

jp112sdl, 2020-07-12:
> Kann ich dafür einen Branch erstellen ?

Das geht beim Wiki leider nicht.

Hab deine Anmerkungen mal ergänzt

JurgenO, 2020-07-12:
Danke für alles ! Ich mach den Issue jetzt zu.

closed 2020-07-12 by JurgenO