jp112sdl/AskSinAnalyzer #69: [regression] ccu_create_devlist.txt: HmIPCentral system.Exec broken after commit b325ebbe1c17766bb87b4b31e1f8c595fcc4af83

Yannik, 2020-07-03:
Hi,

your commit https://github.com/jp112sdl/AskSinAnalyzer/commit/b325ebbe1c17766bb87b4b31e1f8c595fcc4af83 unfortunately breaks correctly setting `HmIPCentral` in the `ccu_create_devlist.txt` script for me. 

After the [system.exec](https://github.com/jp112sdl/AskSinAnalyzer/blob/b325ebbe1c17766bb87b4b31e1f8c595fcc4af83/additional/ccu_create_devlist.txt#L62), `stdout` is empty for me.

Reverting the commit fixes this.

jp112sdl, 2020-07-03:
Hi, which firmware version is your ccu (ccu2 / 3 / raspberrymatic) running at?

Yannik, 2020-07-03:
3.51.6

jp112sdl, 2020-07-03:
Hmm...
Does `WriteLine(stderr);` show anything?

Yannik, 2020-07-03:
It does:

`/bin/sh: 1: arithmetic expression: expecting EOF: "16#B2531A"`

jp112sdl, 2020-07-03:
Please try
```
system.Exec("echo $((0x`grep Adapter.1.Address /etc/config/hmip_address.conf|awk -F'\=' {'print $2'}`))", &stdout, &stderr);
```

Yannik, 2020-07-03:
That works fine.

jp112sdl, 2020-07-03:
So I can put this in this script.

Thanks!

commit 91ac3aca (2020-07-03): fix #69
closed 2020-07-03 by jp112sdl